p2r3/bareiron #38: Build fail idk why

Randomised-gamedev, 2025-09-13:
Microsoft Windows [Version 10.0.26100.6584]
(c) Microsoft Corporation. Med ensamrätt.

C:\Users\name\OneDrive\Skrivbord\bareiron>node build_registries.js
Error: ENOENT: no such file or directory, scandir 'C:\Users\\OneDrive\Skrivbord\bareiron\notchian\generated\data\minecraft'
    at async Object.readdir (node:internal/fs/promises:948:18)
    at async scanDirectory (C:\Users\shaki\OneDrive\Skrivbord\bareiron\build_registries.js:189:17)
    at async convert (C:\Users\name\OneDrive\Skrivbord\bareiron\build_registries.js:325:22) {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'scandir',
  path: 'C:\\Users\\name\OneDrive\\Skrivbord\\bareiron\\notchian\\generated\\data\\minecraft'
}

C:\Users\\OneDrive\Skrivbord\bareiron>

Randomised-gamedev, 2025-09-13:
does not work

Randomised-gamedev, 2025-09-13:
here some more C:\Users\\OneDrive\Skrivbord\bareiron>node node build_registries.js
node:internal/modules/cjs/loader:1148
  throw err;
  ^

Error: Cannot find module 'C:\Users\\OneDrive\Skrivbord\bareiron\node'
    at Module._resolveFilename (node:internal/modules/cjs/loader:1145:15)
    at Module._load (node:internal/modules/cjs/loader:986:27)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:174:12)
    at node:internal/main/run_main_module:28:49 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}

Node.js v20.15.1

C:\Users\\OneDrive\Skrivbord\bareiron>node build_registries.js

p2r3, 2025-09-13:
In the first report, it looks like you haven't dumped the registries from the vanilla server correctly. Follow the README.

In the second report, you just typed the command wrong.

closed 2025-09-13 by p2r3

Randomised-gamedev, 2025-09-13:
> In the first report, it looks like you haven't dumped the registries from the vanilla server correctly. Follow the README.
> 
> In the second report, you just typed the command wrong.

K ill do that