p2r3/bareiron #39: Add Nether?

Rayray1099, 2025-09-13:


p2r3, 2025-09-13:
Maybe at some point, but not as a dimension - that would be much too compute/memory intensive. I'm thinking of something akin to the [Nether Reactor](https://minecraft.wiki/w/Nether_Reactor_Core) from the old Pocket Edition versions. Essentially, a structure/block that converts nearby terrain to Nether blocks and spawns nether mobs.

fox3000foxy, 2025-09-13:
I may start it if this is okay

p2r3, 2025-09-13:
@fox3000foxy This is a pretty big feature request. To be blunt, from what I've seen in your contributions, you've still got a lot to learn about lower-level development before operating in a codebase like this one. You're welcome to try, of course - I'm just warning you that the review process might feel more scathing than you expect.

hew02, 2025-09-14:
Would love to hack on this codebase, is the nether reactor the way to go?

p2r3, 2025-09-20:
Is that not just swapping grass with netherrack and water with lava? That's effectively the same "conversion" I was talking about. There's no need for spread, that would be infeasible given the compute constraints.

I don't think there's anything more to discuss in this issue, someone just needs to assign themself and PR.