p2r3/bareiron #42: Prebuilt versions for the ESP32-C3?

LostShadowGD, 2025-09-13:
Idea:

There should be prebuilt versions for the ESP32C3, with your recommended settings, an MOTD of "An ESP32 server!", and 4 builds for "Peaceful", "Easy", "Normal" and "Hard".

p2r3, 2025-09-13:
I think this is rather arbitrary, and rather tangential to the goal of the project.

LostShadowGD, 2025-09-13:
I just feel it would make it easier for beginners to set up one of these servers for themselves, and then move on to bigger microcontroller projects 😁

p2r3, 2025-09-13:
A pre-built binary is hardly a "project". It's a product. If your goal is to tailor to beginners that want to explore embedded programming, there's no better way to do that than to compile the code yourself. Prebuilt binaries won't teach anyone anything about how it really works.

The reason I say it's tangential to the goals here is because this is a _code_ repository for development. GitHub typically isn't where you find nice pre-packaged consumer apps, it's where you find source code. If you want to upkeep binary releases, that's fine by me, but I don't think this is the place. Singling out the ESP32-C3 specifically is also odd. I get that's what I used in my video, but I hope you can see why that's the "product" part, and this is the "development" side.

Mini rant: I've noticed that most of the issues posted here are assuming that this is a finished product, only offering QoL improvements. I suppose the messaging took a wrong turn somewhere.