p2r3/bareiron #44: Run-length encoding?

max1220, 2025-09-13:
I think it could be possible to compress the block-change list more efficiently using something like run-length encoding.
For this to work properly, the list would have to be split into separate arrays.
The coordinates array could be compressed with a modified RLE that offsets the position in the array with the array index, so simple increments in block positions can be RLE-compressed as well(otherwise RLE would not compress the block positions, since they are unlikely to be unchanging over a continuous range of block-changes).
This could also make the access faster, since you can efficiently skip over large areas of the block-change list.
It would probably have to be benchmarked and tuned to get the best out of this(how many bits for length encoding etc.). It probably also depends on "playstyle"(certain things like rectangle would compress better, possibly even depending on how you build them).

(Very awesome video BTW! :P)

fox3000foxy, 2025-09-13:
This is indeed an idea that I propose to dig a little deeper into and see if we can talk to the original maintainer of the repo about it.

p2r3, 2025-09-13:
This took a moment to wrap my head around. Particularly because, unlike traditional RLE, this doesn't do anything to optimize a row of blocks. Instead, it... optimizes oddly specific scenarios, like building an L-shape.

Also, I think you misunderstood my point about sorting. I didn't say to avoid sorting, I said it'd have to be _compatible_ with sorting and binary search for the idea in issue #33 to work. This... isn't. If sorted in any coordinate-dependent way, it'd take up exactly as many slots as uncompressed entries. Available space depends entirely on the longest run.