p2r3/bareiron #48: Openning chest disconnects the client

pesman129, 2025-09-13:
Hello.

Every time i try to open a chest it just disconnects me from the server. I run the bareiron on my raspberry pi 5 just for fun and my minecraft install on my pc has fabulously optimized modpack on it but i don't think that somehow affects it because my friend has a vanilla install and it does the same thing. I tried enabling the chests in the globals.h and it still does the same. The error it shows in the logs is Disconnected client -1, cause: 1, errno: 9. Can someone help me?

[image]

charles4090, 2025-09-13:
This is likely due to the fact you're using Fabric. Check the warning in the README.

fox3000foxy, 2025-09-13:
This is likely due to the fact you're using other client than vanilla. Check the warning in the README.

fox3000foxy, 2025-09-13:
> because my friend has a vanilla install and it does the same thing

Can you provide more details about the steps you made ? How can we reproduce that bug ?

pesman129, 2025-09-13:
What details?

p2r3, 2025-09-13:
There's a good chance that the memory tricks used for chests aren't compatible with the ARM processor of the RPi. The README does warn of this:
> If you experience crashes or instability related to chests or water, those features can be disabled with ALLOW_CHESTS and DO_FLUID_FLOW, respectively.

And so does the comment above the chest toggle:
> On some platforms, this could cause bad performance or even crashes during gameplay.

In short, ARM is very strict about memory boundaries for primitives, and chests bend the rules a bit in that regard. I think there's a simple enough fix for this though, but you'll have to wait a bit until I can reproduce and test this.

pesman129, 2025-09-13:
Thanks. I was placing the chest where a stone was before and i tried on a surface where an oak log was it did the same and when i placed it where there was an air originally it didn't crash. Really thank you!