p2r3/bareiron #51: zombies are sometimes invisible

SDFTDusername, 2025-09-13:
sometimes i randomly get damaged by zombies while i can't see them, not sure what causes them

p2r3, 2025-09-13:
This is a known bug, and unfortunately a very poorly understood one. At some point I thought this had to do with conflicting UUIDs, though I guess not, as that should've been fixed.

Can you send the log file at `.minecraft/logs/latest.log`?

SDFTDusername, 2025-09-13:
> This is a known bug, and unfortunately a very poorly understood one. At some point I thought this had to do with conflicting UUIDs, though I guess not, as that should've been fixed.
> 
> Can you send the log file at `.minecraft/logs/latest.log`?

[latest.log](https://github.com/user-attachments/files/22315040/latest.log)

p2r3, 2025-09-13:
```
[22:29:21] [Render thread/WARN]: Duplicate entity UUID 53444654-4475-7365-726e-616d65206a6f: class_1452['Pig'/-3, l='ClientLevel', x=4.00, y=69.00, z=-58.00]
[22:29:21] [Render thread/WARN]: Duplicate entity UUID 53444654-4475-7365-726e-616d65206a6f: class_1452['Pig'/-4, l='ClientLevel', x=-48.00, y=66.00, z=-28.00]
[22:29:21] [Render thread/WARN]: Duplicate entity UUID 53444654-4475-7365-726e-616d65206a6f: class_10730['Cow'/-5, l='ClientLevel', x=28.00, y=65.00, z=-28.00]
```
I think that confirms the hypothesis. Very strange, though. I could've sworn that I fixed this. Will look into it.

p2r3, 2025-09-13:
Found the issue, I'm not randomizing the UUID properly on login:
https://github.com/p2r3/bareiron/blob/61eb38a83dbccd26a505fdcfe3fb8e9769e99436/src/main.c#L143-L147

Should be a simple fix.

commit e81e75a6 (2025-09-13): ensure mob uuid uniqueness on login

Closes #51
closed 2025-09-13 by p2r3