p2r3/bareiron #58: Disconnecting from server

AlexJiang124, 2025-09-13:
When I connect to my ESP32 (Nodemcu ESP32-S), after a few seconds, it says Internal Exception: java.net.SocketExeption: Connection reset. In some rare cases, like first flash, it will work for a while, until eventually crashing and saying this in serial:listen failed: Address already in use ; abort was called at PC 0x400f4c5a on core 0. I flashed it with PIO and no changes, only boosting clock to 240mhz. Some notable things I also saw were Unknown packet: 0x07, length:23, state:4, and Unknown packet: 0x00, lenght:5, state:5. Thanks for the help.

AlexJiang124, 2025-09-13:
Also, the server jar is the official 1.21.8 jar

p2r3, 2025-09-13:
Are you using a vanilla client?

AlexJiang124, 2025-09-13:
Yes

p2r3, 2025-09-13:
"Address already in use" makes it sound like the server is trying to restart. I think your internet connection might be bad.

The program attempts to start the server whenever it connects to WiFi and gets an IP. If it completely disconnects and has to reconnect, it'll attempt to start the server again and fail if the port is already taken.

AlexJiang124, 2025-09-13:
Sure, Ill try that, but that only occurs after a while, it doesn't occur every time i try to connect.

AlexJiang124, 2025-09-13:
Actually, moving closer did help, like I might just have to try a few times for it to work. There might be some underlying issue, but I don't know. Thanks for helping, and this is very impressive.

reykato, 2025-09-13:
I may be having a similar issue. I only get "Connection reset" and it happens rather consistently. I've tried running it on an ESP32-WROOM dev board, a Seeed Studio XIAO ESP32S3 (with external antenna attached), and a cheap ESP32-C3 super mini dev board. All of them exhibit the same issues despite placing them directly next to the router and having approximately -30 RSSI. I have tried using another router (mobile hotspot from my PC) to no avail, always the same thing. I've been able to play the game for a few seconds and load a few chunks more than the initial nine, but it always stops loading and shortly after gives me the "java.net.SocketException: Connection reset" output on the client. I can never reconnect to the server after this happens without resetting the chip.

The serial output shows no bad packets. Using the C3 board I occasionally got a task watchdog reset error like this one, but nothing else on the other chips:
`Client Acknowledged Configuration

E (84604) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (84604) task_wdt:  - IDLE (CPU 0)
E (84604) task_wdt: Tasks currently running:
E (84604) task_wdt: CPU 0: bareiron
E (84604) task_wdt: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x40389776  RA      : 0x403897a0  SP      : 0x3fcc6700  GP      : 0x3fc90e00  
TP      : 0x3fca68e0  T0      : 0x00000000  T1      : 0x00000008  T2      : 0x00000000  
S0/FP   : 0x3fcbaea8  S1      : 0x00000000  A0      : 0x00000001  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0x00000005  A4      : 0x3fcb7000  A5      : 0x600c2000  
A6      : 0x60023000  A7      : 0x000000ff  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x3fcc6818  S5      : 0x3fcc681c  S6      : 0x0000000c  S7      : 0x00000005  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000042  T4      : 0x00000506  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x3fcbaea8  MTVAL   : 0x403871ce  
MHARTID : 0x00000000 `

reykato, 2025-09-13:
Vanilla client, yup

Kevin-Dohmen, 2025-09-13:
I have the same issue on my ESP32-S3 (`seeed_xiao_esp32s3`) but on my ESP WROOM 32 (`esp32doit-devkit-v1`)

Kevin-Dohmen, 2025-09-13:
The issue does seem to be less on my WROOM 32
but when standing still for a few seconds i get the connection reset error

reykato, 2025-09-13:
I wasn't entirely accurate saying there were no unknown packets. Seeing that OP had the same packets but was able to fix it by moving closer to their router, I'm not confident these unknown packets have anything to do with my (and now Kevin's) issues.

```
Sending Login Success...

Client Acknowledged Login

Received Plugin Message:
  Channel: "minecraft:brand"
  Brand: "vanilla"

Received Client Information:
  Locale: en_us
  View distance: 12
  Chat mode: 0
  Chat colors: on
  Skin parts: 127
  Main hand: right
  Text filtering: on
  Allow listing: on
  Particles: 0

Sending Server's Known Packs

Sending Registries

Sending Tags

Sending Plugin Message

Received Client's Known Packs
  Finishing configuration

Unknown packet: 0x07, length: 23, state: 4

Packet Contents: 01 09 6D 69 6E 65 63 72 61 66 74 04 63 6F 72 65 06 31 2E 32 31 2E 38
Client Acknowledged Configuration

Unknown packet: 0x00, length: 5, state: 5

Packet Contents: FF FF FF FF 0F 
Unknown packet: 0x00, length: 5, state: 5

Packet Contents: FF FF FF FF 0F 
Sending new chunks (-1, 0)
Generated 5 chunks in 2640 ms (528.00 ms per chunk)

(the server drops connection/hangs here)
```

p2r3, 2025-09-14:
The missing packets look like a perfectly normal login sequence, it's just that bareiron doesn't implement the ones that it doesn't need. There seems to be some much lower-level issue here.

The performance of that initial chunk load is abysmal for an S3 (or even a C3), that could maybe have something to do with it? I don't think that's it, but just to entertain me, could you try setting `TASK_YIELD_TICKS` to `5` and `TASK_YIELD_INTERVAL` to like `250000` in `globals.c`?

reykato, 2025-09-14:
I had to drop VIEW_DISTANCE to 1 to get back into the game without being booted immediately. This is with TASK_YIELD_TICKS at 5 and TASK_YIELD_INTERVAL at 250000

```Sending new chunks (0, 1)
Generated 3 chunks in 3240 ms (1080.00 ms per chunk)
Sending new chunks (1, 1)
Generated 3 chunks in 1790 ms (596.67 ms per chunk)
Sending new chunks (1, 0)
Generated 3 chunks in 1820 ms (606.67 ms per chunk)
Sending new chunks (2, 0)
Generated 3 chunks in 1880 ms (626.67 ms per chunk)
Sending new chunks (2, 1)
Generated 3 chunks in 1750 ms (583.33 ms per chunk)
```

p2r3, 2025-09-14:
Just a hunch, but have you tried a different power supply? I'm thinking of possible environmental factors that could contribute to the scale of this issue, because at this point I really doubt it's the code.

Kevin-Dohmen, 2025-09-14:
I did have power issues with my WROOM D1 Mini (first esp i tried), did some debugging and whenever `esp_wifi_start();` was called it rebooted, tried changing the transmit power, but didn't help.

Power suply did significantly improve world gen speed for me but sometimes i still get kicked when standing still, but takes much longer.

Kevin-Dohmen, 2025-09-14:
@p2r3, power saving did not change anything on my XIAO ESP32-S3 (without custom compiler flags), even tried max power saving `esp_wifi_set_ps(WIFI_PS_MAX_MODEM);`.
Changing cpu clock didn't help either.

p2r3, 2025-09-14:
That's not going to be related to this issue, no. That's just how the server works. It's single-threaded. If the issue was a _task_ not getting enough CPU time, it'd have to be the idle task, and that'd cause the watchdog to visibly trip.

Kevin-Dohmen, 2025-09-14:
but that would mean that if the world is generating it could time out the client right?

p2r3, 2025-09-14:
I mean, sure? If the chunk update takes _over 15 seconds_ to do, yes. But as I said, that's not related to your issue. Unless you're doing something _very_ wrong, a row of chunks will not take that long to generate.

AlexJiang124, 2025-09-14:
I did ask GPT and it said something about the keepalive not being handled correctly, but it never gave any working solutions. Sometimes the server would not reset until I stopped moving, and I couldn't pick up blocks in that state, too.

p2r3, 2025-09-14:
> I did ask GPT and it said something about the keepalive not being handled correctly

Please don't try to use an LLM to debug an issue like this in a codebase like this, you'll only confuse yourself further. Without a ton of additional context, it's prone to proudly claim that it's found the solution when really it doesn't know much about the issue at all. The keepalives are handled fine. Maybe the busy-loop is an arguable choice, but there's no reason not to do so on a microcontroller.

> After some more testing i did notice when the terrain stops generating it takes about 15 seconds before i get the connection reset error

Yes, that's how cause and effect work. If your car's "check engine" light turns on due to the engine being on fire, you don't start blaming the light for the fire, do you? Analogously, if the server freezes up and the chunks stop loading, I feel like the core issue is that the server froze in the first place, not that you were kicked 15 seconds after that.

p2r3, 2025-09-14:
Given that switching power supplies seemed to help some, and we don't have any other leads, I'm going to close this issue as stale for now. @AlexJiang124 try switching to a better, dedicated power supply. I seriously doubt there's a code change that'll fix this, let alone that ChatGPT will be the one to discover it.

closed 2025-09-14 by p2r3