p2r3/bareiron #69: Feature Idea: Creepers

Bro-186, 2025-09-14:
Hi,
First off, i wanted to thank the devolpement team for the energy they putted into this project.
I was thinking that creepers are a central aspect into the minecraft gameplay and it isn't a true minecaft gameplay without them, I tought that you maybe didn't added them because it would take a lot of CPU calculations, so I putted some thinking into it, and I got an Idea what if the creeper's hole was already pre-tederminated, I was thinking that the creeper hole could have a shape similar to the one in the Image, let me know if my idea is garbage or if it could be added

[image]

ccuser44, 2025-09-14:
Actually creepers are probably the most easiest mob to add if you exclude the already added mobs. Just check when attacking if the mob type is a creeper and if it is then create an explosion.

Also I don't see why it would make sense to have a fixed explosion as an optimization. I'd much prefer to have a vanilla style randomized explosion.

The hardest part about creepers is the fuse timer. An instantly exploding creeper would probably be too op

ccuser44, 2025-09-14:
> and a simple sphere of block changes could be used for explosions (compare projected distance squared to radius squared).

I think a small addition should be made to your proposed explosion logic in where the radius and a rng based on distance is checked to have a more rougher shaped explosion like in vanilla