p2r3/bareiron #78: Smooth entity movement

Hahahaloserbozo69, 2025-09-14:
I know that by default to save on resources, the entity movement is stuttery, but could you add an option inside of globals.h that lets you have smooth movement so I don't have to dig through the code? And if I must look through the code, where should i look to make the mob movement smooth?

fox3000foxy, 2025-09-14:
Parametrable mouvement ? Sounds good, but idk if p2r3 will approve the idea.

Hahahaloserbozo69, 2025-09-14:
what should i do if i just want to add that (i mean what .c/.h file should i look in to add it myself) and if you know, what function should i edit/add to make that happen?

Dodi55, 2025-09-15:
lll

TecoTurtle, 2025-09-19:
> This isn't something that you just change a few lines for, unfortunately. The main things preventing mob movement from being truly "smooth" are tickrate and data types.
> 
> The server runs at 1 tick per second by default, though this _can_ be easily changed in `globals.h`. Though if you do this, you'll notice that movement is still just as choppy as before.
> 
> This is because, fundamentally, the server cannot subdivide movement within a block. Coordinates are stored as integers - that's a memory-saving measure, which the readme states as the number one priority. Even if you changed positions to floats and increased the tickrate, you'd then have to rewrite all of the mob movement code, as it's all exclusively integer based. This steps on the second highest priority of this project, which is performance. Doing floating point operations for movement and collisions is not only messier, it's also more computationally expensive. Doing all that for just making mob movement a bit prettier sounds counterproductive to the goal of this project.
> 
> The only approach that I'd consider is some kind of interpolation. But, since the client doesn't provide anything of the sort, it'd have to be done on the server. It _should_ be possible to make that an opt-in feature, at the cost of additional memory and more operations during tick handling. Effectively, you'd be storing each mob's current position and destination separately, and interpolating between them over a second.

You could make a animation for mobs moving between blocks like you did with items

p2r3, 2025-09-20:
That just isn't how that works. Please don't _guess_ solutions if you don't know the code.

TecoTurtle, 2025-09-27:
> That just isn't how that works. Please don't _guess_ solutions if you don't know the code.

Mb I thought it would work because items are entitys