p2r3/bareiron #80: Yet another block changes suggestion (Chunk lists)

caszuu, 2025-09-14:
Currently every time the world state needs to be queried, the entire `block_changes` array must be iterated, even when querying a small area of the world. This load can add up quick if there are already a lot of changes and a lot of chunks are being generated. 

The idea is to replace the world-global `block_changes` array with a set of chunk-local linked lists which are slowly allocated from a global *buffer* as needed. Every time a new chunk is interacted with, its "header" (chunk coords + ptr to the first block change) is appended to the start of the *buffer* and its child block changes are slowly pushed from the end of the *buffer* in batches and linked to one another.

When the server needs to check the state of a block, it can simply find its owning chunk and walk though its block change batches, skipping other chunk changes in the process.

This both improves performance quite a bit (mainly for worldgen) as only specific chunks can be scanned for changed blocks **and** in some cases can marginally reduce memory usage as `BlockChanges` now only store chunk-local coords which fit into 16 bits, cutting `BlockChange` size by half.

The downside of this approach is how intrusive it is in its current form, especially since it will probably not be compatible with other work already being done on block changes, so i'm not sure if its the right fit for this project. Was fun working on it anyway ^^

---

A proof-of-concept is already usable on my [fork](https://github.com/caszuu/bareiron/tree/diff-rework) for benchmarking but neither chests or serialization are implemented yet.

p2r3, 2025-09-14:
I had heard of this chunked idea in YouTube comments ad nauseam, and I had always just suspected it'd be a terrible idea due to having to frequently defragment. Hadn't even considered a linked list.

The idea sounds nice, if done right. I haven't looked at your implementation yet, I'll do that as soon as I get the time to.

caszuu, 2025-09-15:
Well fragmentation on the buffer scope is a non-issue as both structs are allocated in a stack-like way. When allocating they're simply pushed onto it and deallocation uses a swap-and-pop style freeing making sure no gaps appear.

A bigger issue is with low-change chunks as the current implementation reserves changes in batches of 8 (configurable in `global.h`). This helps to reduce the link memory overhead but also wastes memory if batches are not fully filled up. For example a chunk with a single change takes up 40 bytes (12B for header + 28B for initial batch) instead of 6.

This levels out already at 7 changes, but since many chunks will not have many changes (eg. chunks from mining) it's a bit of a problem to be solved.

p2r3, 2025-09-15:
Yeah no, I recognize that fragmentation is not an issue here, that's what I was referring to with my surprise at a linked list.

I'm really not a fan of the overhead though. Considering that our priorities are memory first and performance second, I really wouldn't want to compromise the latter for the former. I'd much sooner consider a data _compression_ proposal (e.g. RLE), or even a performance optimization that doesn't impact memory (e.g. binary search), than one that improves performance with a memory penalty.

caszuu, 2025-09-15:
That's fair. Although i should add that in almost any other case more memory is saved than wasted, if that makes sense.

While chunks have a upfront cost of 12B, they also grow only at effective 3.5 bytes per block (always reserved in 28B blocks) which eventually hides the overhead (both from links and unused entries) as changes are added to a specific chunk.

For reference global changes uses `n * 6` bytes while a *single* chunk uses `12 + (3 * 8 + 4) * ceil(n / 8)` bytes of storage.

So it really depends on the play style of the players, things like mining individual ore blocks will be less efficient, while clearing out larger areas, strip mining or building will be more efficient.