p2r3/bareiron #82: PlatformIO documentation for building to ESP32 target is somewhat ambiguous, and doesn't exactly compile

marrowbuster, 2025-09-14:
Trying to compile this for a Seeed ESP32-S3 XIAO board and see what I can do to increase performance given that the S3 is much beefier than the C3 shown in the video.

I followed the installation guide to the best of my ability, including ensuring I was using ESP-IDF, and cloning this git repo on top of the PlatformIO project folder, and manually extracting the registries from the server jar file, but the first issue that arose was that using `deno` didn't work unless I changed the `build_registries.js` file into a `.cjs` file.

Furthermore in `globals.h`, `ESP_PLATFORM` didn't seem to be `#define`d in my SDK and when trying to build via `pio run` in the command line (as I had to guess what to do since it wasn't specified), there were the errors:

```
fatal: not a git repository (or any of the parent directories): .git
CMake Error at .pio/build/seeed_xiao_esp32s3/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /home/marrowbuster/Documents/PlatformIO/Projects/esp32s3server/.pio/build/seeed_xiao_esp32s3/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /home/marrowbuster/.platformio/packages/framework-espidf/tools/cmake/third_party/GetGitRevisionDescription.cmake:83 (include)
  /home/marrowbuster/.platformio/packages/framework-espidf/tools/cmake/third_party/GetGitRevisionDescription.cmake:93 (get_git_head_revision)
  /home/marrowbuster/.platformio/packages/framework-espidf/tools/cmake/project.cmake:719 (git_describe)
  CMakeLists.txt:3 (project)
```

On EndeavourOS which is Arch-based. If more info is needed, please let me know.

p2r3, 2025-09-14:
This seems like a PlatformIO usage issue, not really related to the project. When using the VScode IDE, you'd just click the arrow button to compile and upload. Not sure about `pio run` or anything, I'm not that familiar with it myself. The internet can probably answer these questions just fine(?)

Svahnen, 2025-09-15:
I do agree that it was harder then I thought getting it to work with PlatformIO. What I did was to firstly clone the repo then setup a new PlatformIO project using PlatformIO extention for VSCode, then copy the content from the repo into the new project, including the hidden .git folder so I later can do a git pull to get updates. After that I had to suppress some warning using the ini file as follows.

platformio.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200

build_unflags =
  -Werror
  -Werror=all

build_flags =
  -Wno-error=parentheses

; This is optional for persistant storage
board_build.filesystem = littlefs
lib_deps =
  https://github.com/joltwallet/esp_littlefs.git
board_build.partitions = max_littlefs.csv
```

Also here is the guide I followed for how to setup LittleFS https://github.com/joltwallet/esp_littlefs, after making it work I increased the littlefs partition, here is my working example (dont forget to update the CMakeLists.txt file as mentioned in the guide):
max_littlefs.csv:
```
# Name,    Type, SubType,  Offset,    Size,      Flags
nvs,       data, nvs,      0x9000,    0x6000,
phy_init,  data, phy,      0xF000,    0x1000,
factory,   app,  factory,  0x10000,   0x100000,
littlefs,  data, littlefs, 0x110000,  0x2F0000,
```