p2r3/bareiron #85: Segfault on macOS x86_64

Un1q32, 2025-09-15:
This happens on my 2006 MPB running 10.6.8, and my 2014 MPB running 14.7.8

[image]

p2r3, 2025-09-15:
What are you doing in-game when this happens?

Unfortunately, I don't own a Mac, so I don't have the means to debug this.

p2r3, 2025-09-15:
I'm guessing this is the rolling Cosmopolitan release binary?

Un1q32, 2025-09-15:
Also my first time joining the server didn't seem to send packets back, so i couldn't get items or load new chunks, fixed on restart though.

p2r3, 2025-09-15:
I'm not really sure how to help you then. Since the release binary worked and you're doing manual native compilation instead, it's likely an issue with some part of your compilation process. The line in your error report should never trip a segfault if the code is compiled correctly, as that array access is gated by the size of the array.

Since the issue is likely not related to the code of this project specifically, and is more likely to be an artifact of your compilation process, I'll be closing the issue.

closed 2025-09-15 by p2r3