p2r3/bareiron #109: Feature request: Slow block ticking loop for saplings

embermctillhawk, 2025-09-16:
**Describe the feature:**
Every tick, check a few (16?) block_changes for potential sapling growth.

**Explain why this feature is necessary:**
Passive sapling growth would be nice.

**Are you planning on implementing this yourself? If so, explain the approach you intend to take:**
I was inspired by [this comment](https://github.com/p2r3/bareiron/issues/33#issuecomment-3287374480) about moving chests to the end of the array, came up with the following pseudocode.

`globals.h`
> #define BLOCKS_PER_TICK 16
> extern int current_block_tick;

`procedures.c handleServerTick`
> if ( current_block_tick >= MAX_BLOCK_CHANGES ) current_block_tick = 0;
> for ( int i = BLOCKS_PER_TICK ; i > 0 ; i-- ) {
> &nbsp;&nbsp;&nbsp;if ( current_block_tick >= block_changes_count ) {
> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;current_block_tick += i;
> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;break;
> &nbsp;&nbsp;&nbsp;}
> &nbsp;&nbsp;&nbsp;if ( block_changes[current_block_tick].block == B_oak_sapling )
> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;placeTreeStructure( block_changes[current_block_tick].x , block_changes[current_block_tick].y, block_changes[current_block_tick].z );
> &nbsp;&nbsp;&nbsp;current_block_tick++;
> }

At default settings, this would cause saplings to grow naturally in about 625 seconds on average.

Without chests being moved to the end of the array, the code would need to be modified to skip over chest data. This would have the side effect of speeding up sapling growth the more chests exist in the world, but only slightly, less than 0.1% per chest at default settings, and that's the worst case scenario of the chest being the last block checked this tick. If it's the first block checked, it may not affect the growth at all.

Or maybe a new variable could be added, `blocks_to_tick`, replacing the local temporary `i`, which is incremented by `BLOCKS_PER_TICK` every tick, and decremented every time a block is checked. If the current `blocks_to_tick` isn't big enough to skip over chest data, then no more blocks are checked this tick, we wait for next tick with its larger `blocks_to_tick`.

Reordering of `block_changes` could potentially cause saplings to be skipped. Not an issue really, adds some more randomness back to tree growth, and I can't think of a sensible workaround.

What may be an issue is if `block_changes` is reordered and puts `current_block_tick` in the middle of chest data... but if chest data is moved to the end of the array, then that can't happen. Or rather, it would already happen all the time that `current_block_tick` passes harmlessly through chest data.

embermctillhawk, 2025-09-16:
Also, new idea. If chests are being moved to the end of the array, then a wholly different strategy for sapling growth ticks is possible that returns the randomness and gets rid of `current_block_tick`.

`globals.h`
> #define BLOCKS_PER_TICK 16
> #define BLOCK_TICK_STEP ( MAX_BLOCK_CHANGES / BLOCKS_PER_TICK )

`procedures.c handleServerTick`
> int r = fast_rand() % BLOCK_TICK_STEP;
> while ( r < block_changes_count ) {
> &nbsp;&nbsp;&nbsp;if ( block_changes[r].block == B_oak_sapling )
> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;placeTreeStructure( block_changes[r].x, block_changes[r].y, block_changes[r].z );
> &nbsp;&nbsp;&nbsp;r += BLOCK_TICK_STEP;
> }

p2r3, 2025-09-16:
Sure. Just gate it with a `#define` in `globals.h`.