p2r3/bareiron #114: Bug report: Global Buffer Overflow in readString

tylzars, 2025-09-16:
A buffer overflow exists in the `readString()` function that can be triggered by sending malformed network packets with oversized string lengths. This can be triggered during the handshake phase by sending a packet with an oversized server address string.

The vulnerability is located in `src/tools.c` in the `readString()` function as there is no bounds check before the data is read into global memory.

An example ASAN error from Linux is below:

```
==645284==ERROR: AddressSanitizer: global-buffer-overflow on address 0x5555555b3120 at pc 0x7ffff787b325 bp 0x7fffffffd6a0 sp 0x7fffffffce48
WRITE of size 1000 at 0x5555555b3120 thread T0
```

On embedded systems, this can be turned into Remote Code Execution. On Linux and Windows it allows someone to overwrite some game state and potentially crash the server.

The following PR implemented a potential fix: [Link](https://github.com/p2r3/bareiron/pull/113).

coremaze, 2025-09-16:
I haven't tested these, but any other instance of this pattern likely has the same impact and would need the same remediation. Here's the first two examples I found: 

https://github.com/p2r3/bareiron/blob/ba86dfd927b5e2432be797e12095642dc4091fe1/src/packets.c#L693-L697

p2r3, 2025-09-17:
While it might seem controversial to say this for a buffer overflow fix, I think it's ultimately not _that_ critical in the context of this server. A system is only as strong as its weakest link, and bareiron has many weak links, often by design.

Sending a malformed packet and risking a segfault or arbitrary code execution _would_ be a serious issue in most other contexts, but there are plenty of reasons you shouldn't host this server for someone who'd do something like that, anyway. It's trivial to cheat and exploit this server. It trusts everything you do, and due to the added complexities in state management, I don't think "fixing" that is even on the roadmap.

I suppose this is a cheap check all things considered, and I probably shouldn't have turned down the initial PR, but it's worth pointing out that (at least currently) this doesn't improve the security of the server. You can reopen the PR.

coremaze, 2025-09-17:
I agree that the impact is likely zero, but people do scan the entire public IPv4 address space for Minecraft servers, and the concern for people cheating ingame should be different than the concern for people taking control of the machine. People who aren't you may can this server.

@tylzars If you reopen the PR, consider searching for the other length-prefixed packet structures and applying the same check or building a helper to do so.

tylzars, 2025-09-18:
Due to the remote and unauthenticated avenue of this bug, if someone had the server open to the internet a bad actor could fully take over the device. I agree that the chance is near zero, but I figured since it wasn't much overhead it could help secure others' usage of the server.

I've gone ahead and reopened the PR with some new modifications to handle all use cases for length prefixed data. Feel free to review #113 .

p2r3, 2025-09-18:
Fixed in 3bde6929768adf802fbe57b27d1ef99204bad8cc

closed 2025-09-18 by p2r3