p2r3/bareiron #118: Bug report: Array OOB

embermctillhawk, 2025-09-17:
**Describe the bug**
Placing a chest while too close to the block changes limit will cause out of bounds access.

**To Reproduce**
Write clear step-by-step instructions on how to reproduce the bug from a new world:
1. Alter 19995 blocks.
2. Place a chest.

**Expected behavior**
The chest placement to do nothing, same as normally placing or altering a block.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Minecraft client details:**
- Version: 1.21.8
- Mods: no

**Platform (if desktop):**
 - Operating System: win10
 - CPU: 13400f

**Additional context**
It's a really niche bug, but does need a fix/wontfix before I proceed with some ideas I have.

p2r3, 2025-09-17:
Sure, should be a simple check. Just make sure to account for the width of the chest's data, too.

FloofCat, 2025-09-21:
@p2r3 Are we good to push this PR? Let me know if we need any more changes.

closed 2025-09-22 by p2r3