p2r3/bareiron #120: Performance expectations

Gaprix, 2025-09-17:
What kind of performance should I expect when running this on an ESP32? I have a Wemos Lolin32 board and it can hardly withstand the processing of a single player (constant connection reset), and the second player will not be able to enter at all. Is this how it should be, or is the problem on my side? What were your test results?

Some tech info:
Wemos Lolin32 v1.0.0
Controller: ESP-WROOM-32, 2-core ESP32-D0WDQ6 (240 MHz) processor based on 32-bit Xtensa LX6
520 KB of static RAM for data and instructions

Compile and upload using PlatformIO in Visual Studio Code

Gaprix, 2025-09-17:
internet speed test in repeater mode at the board showed a speed of 6.6mbps incoming and 10mbps outgoing (The result of my WiFi speed test when connected directly from my device: 68mbps incoming, 50mbps outgoing)
As I understand it, this is not enough? Perhaps it's a problem of poor board design?

p2r3, 2025-09-17:
I can't really say.