p2r3/bareiron #127: Feature request: Better broadcast mechanism for player movement

ccuser44, 2025-09-19:
**Describe the feature:**
Only broadcast movement packets directly for the players in the chunk, otherwise only broadcast per tick.
Making it per chunk instead of distance also works more performantly because you don't have to do have an expensive loop & compatison of the players in the world and you can only loop through the players of the chunk, if the chunk stores that data that is.

**Explain why this feature is necessary:**
This kills two birds with one stone. It fixes the player movements being janky while maintaining performance even fir large amounts of players.

**Are you planning on implementing this yourself?**
Uncertain.

p2r3, 2025-09-19:
You're once again making a few critical assumptions about how this works. No, a player's chunk is not stored in their data struct, it'd have to be calculated from their coordinates.

Either way, I don't really see how this optimizes anything? It's a different approach, but not necessarily a better one.

p2r3, 2025-09-20:
Your proposal sounded quite a bit different, and also more convoluted than that.

Again, there's no chunk coordinates in the player data. You'd still have to do a distance check for all players. Not that that's particularly slow, of course. Even so, you'd have to check further than one single chunk - even with the lowest supported render distance, the client is rendering 25 chunks.

There's already an opt-in system for sending updates in regular tick increments, so that's not new. I understand the benefit in doing a distance check and not networking any position updates outside of client render distance, but the way you describe it makes it sound more fluffed and advanced than it really is.