p2r3/bareiron #138: Bug report: Problem with components

StupidRepo, 2025-09-21:
A few days ago, I was working on a modification to change how item handling worked (to allow for stuff like item damage), when I realised that reading prefixed arrays is completely broken.

Here's the breakdown of what happens at the moment (using `cs_clickContainer` as an example):
- Everything is normal up until `readLengthPrefixedData`
  1. The **amount of items** in the array is read. For an example, let's say the **amount** is 1 (for 1 item component).
  2. It will proceed to discard the data using the **amount of items**. The code, however, was written under the assumption that the **amount of items** in the array is actually the array's **length in bytes**, which **it is not!!!!**
  3. The packet will then be read incorrectly, leading to further corruption of packets sent after that.

So, the code was written to try to dispose of X bytes, where **X is just the amount of items in the array** and **not the actual length in bytes**.

I tried to fix this, but due to all [the different component types](https://minecraft.wiki/w/Java_Edition_protocol/Slot_data#Structured_components), you would need to figure out the length of each component in the array and then discard of that final length.

For my fix, I just made it assume any component sent would be `minecraft:damage` (because that was an improvement I was working on at the time) so it would:
1. Read the amount of items in the array
2. For each item, try to read a component network ID ("NID", I'll call it) and check if it's NID is 3 (for `minecraft:damage`). If it is:
3. Dispose of the damage varint value and continue.
4. Do this until all components have been disposed of (however, considering the fact it'll only ever be 1 `minecraft:damage` component, the whole for loop is useless but still...)

But of course, like I said, if you wanted to expand on this in the future then you would need some way to properly figure out the length of each component (perhaps some kind of lookup table where it's like "okay so `minecraft:damage` component = Z bytes", "`minecraft:blablablah` = ... Y bytes", "the total of that is `Z+Y` bytes so discard of that").

So overall, my question is... is this worth fixing? If so, how would we account for all the different component value data types?

p2r3, 2025-09-21:
I recognize that this is a bug, and patching it would of course be a good thing, but may I ask: why is this important?

You said you found this while trying to add item damage. I don't think I'd accept that as a feature, given that it would dramatically restructure the data containers for items and would increase overall memory usage. Similarly, I don't think I'd accept _any_ feature that adds item components for the same reason.

StupidRepo, 2025-09-21:
> but may I ask: why is this important?

Honestly, it isn't but it was something I just wanted to bring to attention because this could affect anything using that array read function/anything that reads arrays.

> it would dramatically restructure the data containers for items and would increase overall memory usage.

Yeah, I was working on a whole ItemStack implementation lol. It would be, of course, put behind a `#define`, and when thinking of memory constraints I did manage to do it in such a way to not exceed the constraints (I can't give exact figures because I decided to delete the fork and try again because a lot of things weren't working which I decided would be worth a whole re-write of the fork).

Just to ask now, would you consider merging a PR that added an ItemStack-like implementation (**providing its behind a `#define` and manage to do it without hurting memory too much?)

p2r3, 2025-09-22:
I feel like that'd potentially diverge behavior too much, so no. Optional features should be small and isolated, especially if they're opt-in.

I think we can agree that the primary benefit to developing a more capable item system would be to allow expanding upon it. However, this becomes a problem if the "more capable" system is opt-in. A whole class of new features "expanding" items would likely target a dependency that is disabled by default. If a user wanted to take advantage of those new features, they'd have no choice but to take the memory impact and toggle the component-based system that the new features are built on.

I think fundamental changes like these make it much harder to maintain a lightweight server. I'd prefer if we kept working under the limitations of the current system, instead of simply adding a "bloat toggle" whenever the existing fundamentals feel too limiting. It's a slippery slope. At some point, we might as well just recreate all of vanilla, and then we're back to square one.

StupidRepo, 2025-09-22:
Yeah, that makes sense tbh. I decided to rework it anyway for fun.

In case you were curious:
| Base Bareiron | ItemStack Bareiron |
| -- | -- |
| [image] | [image] |
(I did a little bit of running around and collected a few items and such)

If you would like to test for yourself, the fork is [here](https://github.com/StupidRepo/bareiron).
<sub>Just don't look at the code too much, it can be a bit messy in some places. Also, chests will eat your items so maybe disable chests lol. Also also, I probably won't touch that fork again unless I decide to one day randomly update it...</sub>